hardwario/lora-modem #27: MAC version 1.1 does not work with TTN v3

janakj, 2022-03-21:
The firmware currently does not seem to work with TTNv3 when LoRaWAN 1.1 is selected. Join works and all the keys seem to be negotiated correctly, but uplinks fail. I can see the uplink in the gateway log, but it probably gets dropped by the LNS. This could indicate an issue with MIC, or some other similar problem.

For now, selecting MAC 1.0.4 in TTN fixes the issue. However, that is not a long term solution. We should eventually have support for MAC 1.1.

Issues with 1.1 were reported in LoRaMac-node before, so it's possible this is an existing issue with the library, or I am not configuring the library correctly.

janakj, 2022-03-25:
Update: It does seem work with TTNv3 and LoRaWAN 1.1, but there are some issues. The Join seems to be processed correctly in LoRaWAN 1.1 mode, but ADR does not seem to work. Also, AT+LNCHECK does not work, i.e., the device does not get the response, although one can see the uplink message in the TTN console. But then an uplink message sent via AT+UTX works and is correctly decoded on the application server. A confirmed uplink sent via AT+CTX also correctly appears on the TTN application server and the payload can be correctly decoded there, but the node does not get the confirmation and keeps retransmitting the confirmed uplink repeatedly.

This seems to hint at some problem with the downlink messages sent from TTN to the device in response to an uplink message. That would explain why ADR, LNCHECK, or confirmed uplink acknowledgements do not seem to work.

janakj, 2022-03-27:
Possibly related: https://github.com/TheThingsNetwork/lorawan-stack/issues/5343

janakj, 2022-03-30:
This turned out to be quite the hunt.

We've discovered a bug in TTN network server v3.18.2 that prevented downlinks from being scheduled in LoRaWAN 1.1 mode. The above-mentioned ticket has a workaround. The issue is scheduled to be fixed in v3.19.1, to be released at the end of April 2022.

With the workaround, downlinks begin to work again, but TTN and LoRaMac-node v4.6.0 are unable to decrypt each other's FOpts headers. Thus, ADR, link check, and other MAC commands do not work with TTN v3.18.2 in LoRaWAN 1.1 mode. It turns out that there is a security flaw in [LoRaWAN Specification v1.1](https://lora-alliance.org/wp-content/uploads/2020/11/lorawantm_specification_-v1.1.pdf) released in 2017 concerning the encryption of the FOpts header. An [erratum](https://resources.lora-alliance.org/technical-specifications/fopts-encryption-usage-of-fcntdwn-errata-on-the-lorawan-l2-1-1-specification) has been proposed that will most likely be included in LoRaWAN 1.1.1. LoRaMac-node (correctly) implements the erratum, but TTN does not. Thus, when used in LoRaWAN 1.1 mode, LoRaMac-node v4.6.0 is incompatible with TTN v3.18.2 or earlier.

Since there is no mechanism in the current version of LoRaWAN that the device could use to discover what minor protocol version or errata the network supports, there is no way for the firmware to detect the incompatibility. Upon decrypting the FOpts header from TTN, LoRaMac-node gets bogus MAC commands and silently discards the whole FOpts header. TTN does the same on their side for uplinks with FOpts.

The only workaround at the moment is to use the firmware in LoRaWAN 1.0.4 mode. Once the LoRa Alliance releases LoRaWAN 1.1.1, TTN will hopefully fix their network server  implementation.

hubpav, 2022-03-30:
What an impressive investigation! Would you like to try out the LoRaWAN behavior altogether with ChirpStack and e.g. MikroTik LoRa gateway?

janakj, 2022-03-30:
Sure, if you are able to test it with ChirpStack, I think that would be useful. It seems ChirpStack implements the erratum, so it should work:
https://github.com/brocaar/lorawan/blob/master/phypayload.go#L904-L947

janakj, 2022-04-01:
Related: https://github.com/TheThingsNetwork/lorawan-stack/issues/5360

hubpav, 2022-04-01:
Yep. We will certainly test it with ChirpStack in the EU868 band. Yet, it would be useful to try that under US915 and other bands. We have observed quite vast differences in behavior in various bands. I also have a chance to test it with ČRA (local LoRaWAN provider).

With our Brazil partner IoT Connecta, we will run tests also with their national LoRaWAN provider. We have some configuration issues (related to channels) with the original vendor firmware implementation in such a setup.

@hubmartin Could you please run the test against TTN on 868 MHz? Thanks.

closed 2022-06-01 by janakj