hardwario/lora-modem #46: AT+VER response

hubmartin, 2022-04-04:
Hi @janakj ,
I'm trying the new firmware v 1.0.0. I noticed that response to AT+VER is:
`AT+VER?<\r>+OK=1.1.06,Mar 31 2022 18:34:02`
(first time I was confused whether I flashed module properly, but the year 2022 confirmed that I really use 1.0.0 ;))

TOWER SDK does not check this string and in case of unknown commands the Murata FW answers with Unknown command `+ERR=-1` which the TOWER SDK handles properly and assumes the firmware without this command is used.

For example https://github.com/hardwario/twr-sdk/blob/master/twr/src/twr_cmwx1zzabz.c#L649 and https://github.com/hardwario/twr-sdk/blob/master/twr/src/twr_cmwx1zzabz.c#L1156

Do you have plans to return in firmware you release as 1.0.0 the same string as a response with AT+VER? I might suggest that we might for example start releasing the firmware with 2.0.0 and assume that AT+VER >= 2.0.0 is our new oss firmware. This version 2.0.0 will also appear on the LCD with the twr-lora-tester firmware.

Thanks

janakj, 2022-04-05:
> Hi @janakj , I'm trying the new firmware v 1.0.0. I noticed that response to AT+VER is: `AT+VER?<\r>+OK=1.1.06,Mar 31 2022 18:34:02` (first time I was confused whether I flashed module properly, but [...]

Yes, I also return `+ERR=-1` in such cases. Whenever you submit an AT command that is not supported, you should get a `+ERR=-1` back.

> Do you have plans to return in firmware you release as 1.0.0 the same string as a response with AT+VER? I might suggest that we might for example start releasing the firmware with 2.0.0 and assume [...]

I have decided to keep AT+VER compatible with the Murata firmware. It returns 1.1.06 to indicate which Murata firmware version the open firmware emulates. That is, the AT+ commands implemented by this firmware emulate Murata firmware 1.1.06.

If you want to know the real version of the open firmware, you can use the command `AT$VER` which returns something like this:
```txt
v1.0.0-3-gba3a6fd3 (modified) [LoRaMac v4.6.0-5-g7d79e82a],2022-Apr-04 01:22:47 EDT
```
To detect which firmware is loaded in a Type ABZ module, you can check whether it implements AT$VER. If it is not implemented, it's the Murata firmware. If it is implemented, it is the open firmware.

The syntax of AT$VER isn't final. I am also thinking of including LoRaWAN version, i.e., which version of LoRaWAN and regional parameters are supported.

janakj, 2022-04-10:
I think it's best to combine AT$VER with AT+VER when you display firmware version on the LCD. That way, we can keep AT+VER bacwards-compatible and at the same time, we can communicate more information to the application via AT$VER.

I think there is nothing to do here on the firmware side, so I am closing the issue. Feel free to reopen if you still think there is a problem.

closed 2022-04-10 by janakj