hardwario/lora-modem #47: Support for AT+RFQ?

hubmartin, 2022-04-04:
I'm trying the 1.0.0 with twr-lora-tester and this firmware uses AT+RFQ? to get the info of last received packet.

If the command is not supported, could you please reply with the `+ERR=-1` like the Murata does? On this reply I ignore errors if the command is not known/supported.
https://github.com/hardwario/twr-sdk/blob/master/twr/src/twr_cmwx1zzabz.c#L1156

Right now it failes on timeout because it seem like it does not reply anything.

```
# 117.10 <D> LoRa TX: AT+JOIN<\r><\r><\n>
# 117.60 <D> LoRa RX: +OK<\r><\r><\n>
# 125.30 <D> LoRa RX: +EVENT=1,1<\r><\r><\n>
$JOIN_OK<\r><\n>
# 125.30 <D> LoRa TX: AT+RFQ?<\r><\r><\n>
# 125.90 <D> $LORA_ERROR<\r><\n>     ##### probably timeout and LoRa Module reinitialization in TOWER SDK
# 128.90 <D> LoRa syncTX: AT<\r><\r><\n>
# 128.95 <D> LoRa TX: AT<\r><\r><\n>
# 129.15 <D> LoRa RX: +OK<\r><\r><\n>
....
```

janakj, 2022-04-04:
Actually, the firmware does support AT+RFQ:
```txt
AT+RFQ?
+OK=0,0
```
So this might be a more general problem with the communication between the firmware and the LoRa modem. We'll have to investigate.

Also, the firmware does respond with +ERR=-1 to all unsupported AT commands. If you don't get the answer, the modem did not detect your AT command, or the answer got lost for some reason.

AT+FRMCNT? is also supported:
```txt
AT+FRMCNT?
+OK=71,39
```

hubmartin, 2022-04-04:
Thanks for details. I'm just starting with TOWER SDK so the timing or communication might be an issue.

I'm seeing some random issues in LoRa communication/JOIN so for now I'll use direct AT commands without TOWER SDK to get rid of the most unknowns in the equation. I'll keep you informed.

janakj, 2022-04-10:
Split into two separate issues: #52 and #53.

closed 2022-04-10 by janakj