hardwario/lora-modem #73: Implement time synchronization

janakj, 2022-06-05:
The LoRaWAN protocol provides a couple of mechanisms that could be used to synchronize the clock on the device. Design an ATCI protocol through which LoRaWAN-based time synchronization could be extended all the way to the main MCU on the Core module. With that, we could synchronize the wall clock time in the RTC via LoRa without the need to have a GPS receiver on the device.

This could be useful in devices such as the LoRa tester where the time could be displayed on the LCD.

vicatcu, 2023-04-19:
I've been wanting this too, for setting the time on an hardware RTC from "network time"

janakj, 2023-04-21:
What accuracy are you looking to obtain? Second-level accuracy is relatively simple. Anything better may require support from the LoRaWAN network.

-Jan

vicatcu, 2023-04-22:
Second-level accuracy is more than adequate adequate for my purposes

janakj, 2023-05-27:
Hi @vicatcu,

I just committed the first version of time synchronization support based on the `DeviceTimeReq` LoRaWAN MAC command. See [this wiki page](https://github.com/hardwario/lora-modem/wiki/Time-Synchronization) for more information. If you know how to build your own firmware binary the source code, feel free to give it a try.

There is a new AT command called `AT$DEVTIME` which triggers time synchronization in the modem. You will eventually get the current time via an `+ANS` notification:
```
AT$DEVTIME
+OK

+EVENT=2,1

+ANS=13,1685213814,746
``` 
Number 13 in the `+ANS` notification identifies a time synchronization response. The second parameter is POSIX time, i.e., the number of seconds since the UNIX Epoch. The third parameters represents fractions of a second. This method should be accurate to +- 100 milliseconds. Please note that POSIX time does not account for leap seconds, i.e., the returned time will be some number of seconds behind UTC time.

This synchronization method requires LoRaWAN 1.0.3 or higher. I have tested it with the The Things Network where it seems to work well.

This feature will eventually be released as part of version 1.5.0.

-Jan

janakj, 2023-05-30:
This feature is now implemented in the firmware, in the companion Python library, and documented in the wiki.

closed 2023-05-30 by janakj