hardwario/lora-modem #107: Not able to join network in US915

jefrin-altrosyn, 2022-09-19:
I have flashed the latest firmware v1.2.4 into an ABZ-078 module in an MKRWAN 1310 board using the SAMD21 MCU as a UART bridge. I modified the Lora network settings of the module for my initial testing using the lora modem python library.
[image]
The module is configured in the US915 region with channels 0 to 7 enabled in the module as well as the gateway. I am using chirpstack as the network server and a multi-tech conduit in packet forwarder mode as the gateway. But the Murata module is not able to join a network. 
The join requests were not even showing up in the network server. When I read the logs in the gateway I saw the messages from the devices failed CRC check.
`JSON up: {"rxpk":[{"tmst":2972878636,"chan":7,"rfch":1,"freq":903.700000,"stat":-1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":-11.2,"rssi":-108,"size":147,"data":"uOIA8JEHg89JheTumbWmVmeSXd4lkI2k+KTFWTnLDJb+w9ejUkiCMIp9EPuxvuPdmd3aSqjjCwaZos/mSLRLhvnf/n+tyBrok0KsW6GDKmWt4WPwsTGkbG2dPHrXNfgqtCJ7dtlvp/ro8pgAHw1NBiVCmRUZC37D7NVhf/8ez2fRw7VZMdkcBidwzT1LJLeCh25C"}]}
INFO: [down] PULL_ACK received in 237 ms`

I can figure out the reason for this issue. I was able to do lora comms fine with the current setup fine previously with the mkrwan firmware & library.

I have modified the following settings so far
1. band - US915
2. mode - OTAA
3. chmask - 0-7 channels
4. adr -false
5. deveui
6. appeui
7. appkey

Am I supposed to modify rf settings other than the above ones in the module?

Any help would be appreciated, Thanks.

hubmartin, 2022-09-19:
Hi, when I cannot see LoRa packets even in the network server, the issue was always in the private/public network configuration. Try to change the Network type (AT command `AT$NWK`) to private. The public is usually for TTN only.

jefrin-altrosyn, 2022-09-19:
Thanks for the suggestion but I am still facing the same issue. I changed the network type to private in the device as well as in the lorawan gateway. I am using a multi-tech conduit as my lorawan gateway.
### Gateway settings
[image]
### packet logs
[image]

I found a similar CRC error issue reported by someone [here](https://forum.chirpstack.io/t/switching-eu868-to-us915-causes-crc-errors-in-cloned-i-cube-lrwan/9989) I am now curious If I should change the rf power or similar settings of the module to something other than the default.

janakj, 2022-09-19:
The settings shown by the Python program seem correct to me. We may need to try to triage this to try to identify the problem.

If you used the original firmware in the device and then flashed 1.2.4 without resetting (removing and readding) the device in ChirpStack, this is the first thing that I would try. The network server and the device have to keep plenty of information synchronized. The original MRKWAN firmware is probably only compatible with LoRaWAN 1.0.2. Our firmware supports LoRaWAN 1.0.4/1.1.1, so try to make sure that the device appears as a completely new device to ChirpStack.

Next question: What LoRaWAN protocol version did you configure for the device in ChirpStack? 1.0 (1.0.4), or 1.1? If you have the device configured as a LoRaWAN 1.1 device, you may need to check that all your keys are correct in the device and in the NS. LoRaWAN 1.1 has a few additional security keys defined: NwkKey (in addition to AppKey), FNwkIntKey, SnwkIntKey, NwkEncKey. You can display all of these keys configured in the device with the Python program.

It is generally easier to start with LoRaWAN 1.0 and then eventually migrate to 1.1. So try to configure ChirpStack to use LoRaWAN 1.0.4 and RP2-1.0.3 for the device in the Device Profile.

Our firmware transmits OTAA Join requests with the maximum RF transmit power and the slowest spreading factor. If your device is very close to the gateway, I can imagine the device might be overloading the gateway's RF frontend, at which point you will most likely receive a corrupted Join message. If you suspect that this might be happening, consider lowering the RF transmission power before sending the Join with [AT+RFPOWER](https://github.com/hardwario/lora-modem-abz/wiki/AT-Command-Interface#atrfpower). Keep the mode (it does not do anything) and keep increasing the index value.

jefrin-altrosyn, 2022-09-22:
Thanks for your quick response and forgive my late response. I tried a few things like adjusting rf power setting and tried older firmware versions like 1.2.3 and 1.2.1 in the device but still no progress. In chirpstack, I am not seeing any lorawan packets not even join requests from the end device in both gateway and device live lorawan frames section.
[image]

> Are you sure that this is message from the device? The Python program shows that you have SF10BW125 configured in the device, but the JSON object shows SF7BW125. This looks like an uplink from another device to me.

Further testing with the device I found that the above CRC failed packets which I thought were from the device were not from the device but from some other unknown source. I saw those packets in the packet forwarder logs even when the end device was offline. Other than the above similar CRC failed packets I am not seeing any packets at all in the packet forwarder logs.

> Technically, you should not need to modify the channel mask before OTAA Join. When transmitting an OTAA Join, the device automatically cycles through sub-bands. In other words, the first transmission happens in 0-7, the second in 8-15, and so on. The NS is then supposed to configure the correct channel mask into the device in JoinAnswer.

  Thanks for the info. I started modifying the channel masks because I did not see any packets from the device in the gateway logs.

I even tried basic station mode in my gateway as well as tried other frequency sub-bands with packet forwarder mode  setting in the gateway and no results.

> If you only configure the appkey (and not nwkkey), make sure that you have LoRaWAN 1.0 selected as the protocol for the device on the network server.

I forgot to mention but I did configure nwkkey in the device an tried joining.

> Our firmware transmits OTAA Join requests with the maximum RF transmit power and the slowest spreading factor. If your device is very close to the gateway, I can imagine the device might be [...]

I moved the device into other rooms far from the gateway but still the same problem. I tried to make the device join the network after modifying the rf power setting with index values from 0 to 14 but no results. My gateway is not picking any packets from the device at all. It makes me think that even if the device is transmitting at all.

> It is generally easier to start with LoRaWAN 1.0 and then eventually migrate to 1.1. So try to configure ChirpStack to use LoRaWAN 1.0.4 and RP2-1.0.3 for the device in the Device Profile.

Since I am in a state where I am not even seeing any join requests in the console. I think I can worry about chirpstack configurations later, but yes I have created device profiles and an entry for the device with settings matching lorawan 1.1 (tried 1.0.x as well).

When I use the mkrwan firmware in the Murata module in the US915_HYBRID band setting I can join the network and transceive messages with my current setup ( i.e multi-tech conduit MTCDT-247A as gateway and chirpstack as network server).

Have you guys ever tried any multi-tech gateways like conduit and had any issues or experiences with those gateways and the Murata module in the US915 region?

janakj, 2022-09-27:
I haven't tried the specific gateway that you mention. I did try a bunch of other gateways including Browan, Mikrotik, and RAKWireless gateways. Never had any issues, as long as RF channels were configured correctly.

As the next step, could you try to load the debug variant of the firmware into the Murata chip and see what it prints to the USART port (the port used to flash Murata) as you perform a Join? The debug variant includes a logger on the USART port which prints a lot of useful debugging information, including all RF parameters configured by the LoRaWAN MAC into the SX1276 radio. If you manage to get the log, please post it here so that we can take a look.

Also, I'd recommend to configure the device as LoRaWAN 1.0 in ChirpStack. The older protocol version is easier to get working (fewer security-related parameters one needs to tweak). You can upgrade to LoRaWAN 1.1 once you get basic communication working.

jefrin-altrosyn, 2022-09-27:
I don't have any other gateways right now at my disposal , but will try to get access to one.

> As the next step, could you try to load the debug variant of the firmware into the Murata chip and see what it prints to the USART port (the port used to flash Murata) as you perform a Join? The [...]

Sure, I will try to load the debug variant firmware into the Murata chip and will keep you posted. 

> Also, I'd recommend to configure the device as LoRaWAN 1.0 in ChirpStack. The older protocol version is easier to get working (fewer security-related parameters one needs to tweak). You can upgrade to LoRaWAN 1.1 once you get basic communication working.

Yea will make sure the device is configured in LoRaWAN 1.0 in chirpstack.

jefrin-altrosyn, 2022-09-29:
After flashing the debug variant firmware into the module, I am not able to interact with the Murata module with SAMD21 MCU as bridge through AT commands.

Firmware flashing process:
1. Flash the fw update bridge code from [here](https://github.com/arduino-libraries/MKRWAN/blob/master/examples/FWUpdaterBridge/FWUpdaterBridge.ino) into the samd mcu then flash the debug build into Murata using the bcf tool.
2. After flashing the murata module with this firmware, Flash the following code into the SAMD mcu for interacting with the murata module from the PC.

```
#define SerialMKRUSB Serial
#define SerialLora Serial2

void setup() {
  SerialMKRUSB.begin(19200);
  SerialLora.begin(19200);

  delay(1000);
  
  pinMode(LORA_BOOT0, OUTPUT);
  digitalWrite(LORA_BOOT0, LOW);
  pinMode(LORA_RESET, OUTPUT);
  digitalWrite(LORA_RESET, HIGH);
  delay(200);
  digitalWrite(LORA_RESET, LOW);
  delay(200);
  digitalWrite(LORA_RESET, HIGH);
}

char rx_buf[512];
char tx_buf[512];

int rx = 0;
int tx = 0;

void loop() {
  while (SerialMKRUSB.available()) {      // If anything comes in Serial (USB),
    tx_buf[tx++] = SerialMKRUSB.read();   // read it and send it out SerialLora
  }

  if (tx > 0) {
    SerialLora.write(tx_buf, tx);
    tx = 0;
  }

  while (SerialLora.available()) {      // If anything comes in Serial (USB),
    rx_buf[rx++] = SerialLora.read();   // read it and send it out SerialLora (pins 0 & 1)
  }

  if (rx > 0) {
    SerialMKRUSB.write(rx_buf, rx);
    rx = 0;
  }
}
``` 

> 2) A dump of all modem’s configuration parameters (can be obtained with "lora get -a”); 3) A list of all channels configured in the modem (can be obtained with "lora channels show”);

I have attached the configuration parameters and active channels of the release build  [here](https://pastebin.com/Hfit5Xqv).

janakj, 2022-09-29:
There is no difference in the AT command interface port configuration between release and debug builds. Are you sure you are flashing the correct firmware?

Try to pass `--erase-eeprom --unprotect --slow` to `bcf flash`. I am not sure these options would change anything, but it's worth a try. 

Also, please try to see if the debug firmware logs anything to the USART port. Load the fw update bridge code into the SAMD21 MCU, flash the debug firmware, **leave the fw update bridge code in the SAMD21 MCU**, run `bcf log` to see message printed to the USART port, reboot Murata.

Debug builds start printing debugging information to the USART immediately after Murata is rebooted. This is what the log looks like:
```
0.00 <I> LoRa Module 1.2.4 [LoRaMac 4.6.0-42-g32534e15] built on 2022-Sep-06 15:36:04 CEST
0.00 <D> part: Opened block 0x20000548 (6144 B), 8 parts of 8
0.00 <D> Invalid system configuration checksum, using defaults
0.00 <D> LoRaMac: Initializing for region EU868, regional parameters RP002-1.0.3
0.00 <D> SX1276SetBoardTcxo: 1
0.02 <D> SX1276SetAntSwLowPower: 1
0.03 <D> SX1276SetAntSwLowPower: 0
0.06 <D> SX1276SetAntSwLowPower: 1
0.06 <D> SX1276SetBoardTcxo: 0
0.06 <D> SX1276SetBoardTcxo: 0
0.07 <D> LoRaMac: Device DevEUI: 323838377B308503 class: A ADR: 1
0.07 <D> LoRaMac: Network public: 1 activated: ABP MAC: 1.0.4 NetID: 00000000 DevAddr: 0362C684
0.07 <D> LoRaMac: Starting
0.07 <D> mlme_confirm: MlmeRequest: 1 Status: 0
```

jefrin-altrosyn, 2022-09-30:
> There is no difference in the AT command interface port configuration between release and debug builds. Are you sure you are flashing the correct firmware?

I flashed  this [lora-modem-abz-1.2.4.debug.bin](https://github.com/hardwario/lora-modem-abz/releases/download/v1.2.4/lora-modem-abz-1.2.4.debug.bin) into the module.

> Also, please try to see if the debug firmware logs anything to the USART port. Load the fw update bridge code into the SAMD21 MCU, flash the debug firmware, **leave the fw update bridge code in the SAMD21 MCU**, run `bcf log` to see message printed to the USART port, reboot Murata.

After flashing the debug build, ```bcf log``` isn't printing any messages to the console.
[image]

The BOOT0 pin is High in the fw update bridge code. Should it be Low ( to load the fw from flash)  for ```bcf log``` to work?
So just in case to test I modified the fw update bridge code with a push button to change BOOT0 pin state to LOW after fw flash as below.
```
#define Serial1 Serial

int button = 2;
int buttonState;

void setup() {

  Serial1.begin(115200, SERIAL_8E1);
  Serial2.begin(115200, SERIAL_8E1);

  delay(3000);

  pinMode(button, INPUT_PULLUP);

  pinMode(LORA_BOOT0, OUTPUT);
  digitalWrite(LORA_BOOT0, HIGH);

  pinMode(LORA_RESET, OUTPUT);
  digitalWrite(LORA_RESET, HIGH);
  delay(200);
  digitalWrite(LORA_RESET, LOW);
  delay(200);
  digitalWrite(LORA_RESET, HIGH);
}

void resetModule() {

  // change murata boot0 pin state

  digitalWrite(LORA_BOOT0, LOW);

  Serial1.println("resetting module");
  digitalWrite(LORA_RESET, HIGH);
  delay(100);
  digitalWrite(LORA_RESET, LOW);
  delay(100);
  digitalWrite(LORA_RESET, HIGH);

  while (!Serial1);
}

char rx_buf[512];
char tx_buf[512];

int rx = 0;
int tx = 0;

void loop() {

  buttonState = digitalRead(button);

  if (buttonState == LOW) {
    Serial1.println("button pressed !!!");
    resetModule();

    delay(200);
  }

  while (Serial1.available()) {      // If anything comes in Serial (USB),
    tx_buf[tx++] = Serial1.read();   // read it and send it out to Murata
  }

  if (tx > 0) {
    Serial2.write(tx_buf, tx);
    tx = 0;
  }

  while (Serial2.available()) {      // If anything comes in Murata
    rx_buf[rx++] = Serial2.read();   // read it and send it out Serial (USB)
  }

  if (rx > 0) {
    Serial1.write(rx_buf, rx);
    rx = 0;
  }
}
```
Still no messages printed in the console.

After modifying the baud rate to 19200 and boot0 to low
```
  Serial1.begin(19200);
  Serial2.begin(19200);
   ....
  digitalWrite(LORA_BOOT0, LOW);
```
After Murata reset with debug builds, I can see the below message in arduino serial monitor by itself. When I issue any  AT commands like AT$VER? the module isn't printing any messages. 
[image]

In the release build, the above message won't be printed but the module will respond to my AT commands.
[image]

With "lora" cli getting response as belows with debug build
[image]

> Try to pass `--erase-eeprom --unprotect --slow` to `bcf flash`. I am not sure these options would change anything, but it's worth a try.

Tried it with release build, can't join the network with release build. With debug build no response as above.

> Debug builds start printing debugging information to the USART port immediately after Murata is rebooted. This is what the log looks like:

Just making sure, For printing the debug logs the Murata module will use its USART2 pins 23,24 and not USART1 19,20 right? (pins 19,20 are left without any connection in mkrwan boards)

janakj, 2022-10-03:
> I flashed this [lora-modem-abz-1.2.4.debug.bin](https://github.com/hardwario/lora-modem-abz/releases/download/v1.2.4/lora-modem-abz-1.2.4.debug.bin) into the module.

Yes, that is the correct firmware binary.

> After flashing the debug build, `bcf log` isn't printing any messages to the console. [image]
>
> The BOOT0 pin is High in the fw update bridge code. Should it be Low ( to load the fw from flash) for [...]

You are right. BOOT0 needs to be low in order for the firmware to start. I didn't realize the fw update bridge code would keep it high, but you seemed to have figured that out on your own below.

> So just in case to test I modified the fw update bridge code with a push button to change BOOT0 pin state to LOW after fw flash as below.
> 
> ```
> #define Serial1 Serial
> 
> int button = 2;
> int buttonState;
> 
> void setup() {
> 
> 
>   Serial1.begin(115200, SERIAL_8E1);
>   Serial2.begin(115200, SERIAL_8E1);
> 
>   delay(3000);
> 
>   pinMode(button, INPUT_PULLUP);
> 
>   pinMode(LORA_BOOT0, OUTPUT);
>   digitalWrite(LORA_BOOT0, HIGH);
> 
>   pinMode(LORA_RESET, OUTPUT);
>   digitalWrite(LORA_RESET, HIGH);
>   delay(200);
>   digitalWrite(LORA_RESET, LOW);
>   delay(200);
>   digitalWrite(LORA_RESET, HIGH);
> }
> 
> void resetModule() {
> 
>   // change murata boot0 pin state
> 
>   digitalWrite(LORA_BOOT0, LOW);
> 
>   Serial1.println("resetting module");
>   digitalWrite(LORA_RESET, HIGH);
>   delay(100);
>   digitalWrite(LORA_RESET, LOW);
>   delay(100);
>   digitalWrite(LORA_RESET, HIGH);
> 
>   while (!Serial1);
> }
> 
> char rx_buf[512];
> char tx_buf[512];
> 
> int rx = 0;
> int tx = 0;
> 
> void loop() {
> 
>   buttonState = digitalRead(button);
> 
>   if (buttonState == LOW) {
>     Serial1.println("button pressed !!!");
>     resetModule();
> 
>     delay(200);
>   }
> 
> 
>   while (Serial1.available()) {      // If anything comes in Serial (USB),
>     tx_buf[tx++] = Serial1.read();   // read it and send it out to Murata
>   }
> 
>   if (tx > 0) {
>     Serial2.write(tx_buf, tx);
>     tx = 0;
>   }
> 
>   while (Serial2.available()) {      // If anything comes in Murata
>     rx_buf[rx++] = Serial2.read();   // read it and send it out Serial (USB)
>   }
> 
>   if (rx > 0) {
>     Serial1.write(rx_buf, rx);
>     rx = 0;
>   }
> }
> ```
> 
> Still no messages printed in the ternminal from `bcf log`.
> 
> When I opened the serial port from arduino serial monitor or putty with baud rate 115200, I am receiving the below response for AT$VER? or other AT commands. [image]
> 
> After modifying the baud rate to 19200 and boot0 to low
> 
> ```
>   Serial1.begin(19200);
>   Serial2.begin(19200);
>    ....
>   digitalWrite(LORA_BOOT0, LOW);
> ```
> 
> After Murata reset with debug builds, I can see the below message in arduino serial monitor by itself. When I issue any AT commands like AT$VER? the module isn't printing any messages. [image]

This is very strange. `+EVENT=0,0` indicates that the Murata modem has been rebooted. That's normal and expected. But `+EVENT=1,1` indicates a **successful** OTAA Join. The modem only generates `+EVENT=1,1` in response to a Join answer received from the network server. I cannot explain how this could be happening here. Two options come to mind: 1) there is a bug in the modem initialization logic that is only triggered on your device, 2) somebody somewhere else is issuing AT+JOIN, e.g., some kind of port mixup.

> Just making sure, For printing the debug logs the Murata module will use its USART2 pins 23,24 and not USART1 19,20 right? (pins 19,20 are left without any connection in mkrwan boards)

I am looking at the Type ABZ datasheet. Pins 41 and 42 are mapped to LPUART1. This is the port used by the AT command interface. The port is configured by default as 19200 8N1.

Pins 18 and 19 are mapped to USART1. This is the port I am using to flash firmware into Murata and to send debugging messages in debug builds. If that port is left unconnected in the MKRWAN board then that might explain why you are not seeing any debugging messages. When used for debugging, the port is configured as 115200 8N1 in the firmware.

Pins 23 and 24 are mapped to USART2. I don't use that port in the modem firmware. If this is the only serial port (other than LPUART1) connected to the host MCU, then I am guessing you are flashing Murata via this port? I didn't know the Murata chip can be also flashed over USART2, but perhaps the STM bootloader supports both USART ports simultaneously.

Anyway, I think we may need to take a look at the debug log to see what's going on. Let me make a custom build for you that will print debugging messages to either USART2 or LUPART1 (or both). That should be a trivial modification. I am hoping to have the build ready for you later this week.

Thanks for all your debugging and investigation effort so far!

jefrin-altrosyn, 2022-10-03:
> Anyway, I think we may need to take a look at the debug log to see what's going on. Let me make a custom build that will print debugging messages to either USART2 or LUPART1 (or both). That should be a trivial modification. I am hoping to have the build ready for you later this week.
> 
> Thanks for all your debugging and investigation efforts so far!

Sure, I can wait for that custom build. Thanks for taking your time to look at this issue.

janakj, 2022-10-05:
@jefrin-altrosyn Here is a debug build of 1.2.4 with logging configured to go to USART2. Please give it a try and let me know if you see any debugging messages. I  haven't tested logging via USART2 since all my Murata chips have that port disconnected.

[firmware.bin.gz](https://github.com/hardwario/lora-modem-abz/files/9717520/firmware.bin.gz)

janakj, 2022-10-05:
@jefrin-altrosyn Somebody else just reported a very similar issue and upon looking at the MKRWAN1310 schematic, I realized that this board uses a different GPIO pin to control the SX1276 oscillator. This might explain why you can't get the modem to transmit. Please find attached a modified version of 1.2.4 for MKRWAN1310. Could you, please, give this a try first to see if it fixes your issue? This is a release build (no debug messages).

[lora-modem-abz-1.2.4.TCXO-PB6.bin.gz](https://github.com/hardwario/lora-modem-abz/files/9718897/lora-modem-abz-1.2.4.TCXO-PB6.bin.gz)

jefrin-altrosyn, 2022-10-06:
> @jefrin-altrosyn Somebody else just reported a very similar issue and upon looking at the MKRWAN1310 schematic, I realized that this board uses a different GPIO pin to control the SX1276 oscillator. This might explain why you can't get the modem to transmit. Please find attached a modified version of 1.2.4 for MKRWAN1310. Could you, please, try this first to see if it fixes your issue? This is a release build (no debug messages).
> 
> [lora-modem-abz-1.2.4.TCXO-PB6.bin.gz](https://github.com/hardwario/lora-modem-abz/files/9718897/lora-modem-abz-1.2.4.TCXO-PB6.bin.gz)

Thank you so much @janakj  for your efforts and help. Flashed this firmware and now the device is able to join the network and transceive messages.
[image]

jefrin-altrosyn, 2022-10-06:
> @jefrin-altrosyn Here is a debug build of 1.2.4 with logging configured to go to USART2. Please give it a try and let me know if you see any debugging messages. I haven't tested logging via USART2 [...]

Tried this debug firmware but I am not seeing any debug messages.

After module reset, I am receiving below message in serial monitor and ```bcf log``` is not returning anything as before. 
[image]

janakj, 2022-10-06:
Great. I am glad the modified version works for you. Starting with the next version, I will make binary releases with the correct settings for MKRWAN boards.

commit 568018e0 (2022-10-13): Add release variants for Arduion MKRWAN boards

Arduino MKRWAN boards only have USART2 connected, so the debugging
logger needs to be redirected there. MKRWAN1310 boards use PB6 to
control TCXO IC power instead of the default PA12.

Closes #109
Closes #108
Closes #107