hardwario/lora-modem #108: Use with MKR1310 - no radio signal

disk91, 2022-10-05:
Hello, 

I've been successfully able to flash the last release of the firmware on the Arduino MKR1310 board. The AT commands works perfectly and I can setup the devEUI as expected.

But When trying to Join, It failed (no response from LNS), The LNS did not received the Join Requets.
My investigations conducted to no radio transmission during the JOIN phase ( flat spectrum analyzer on 867-869) 

Any idea ?

janakj, 2022-10-05:
Have you switched the modem to OTAA mode with `AT+MODE=1`? It's in ABP mode by default after factory reset.

disk91, 2022-10-05:
Yes I did it.

I did a test with a MKR1300 and I've got a radio signal received by LNS.
But with my MKR1310 no radio signal. 

Do you know if anyone, on last release, had a success in using it on a MKR1310 on EU868 ? (in that case it could be a board failure ... I'll double check with the standard firmware also to see if there is a difference).

janakj, 2022-10-05:
Glad to hear the firmware worked for you on a MKR1300! You'll probably the first to make it work on a MKRWAN board.

Regarding MKRWAN1300, I am looking at the schematic and I just noticed that this board has the VDD_TCXO pin (pin 48) wired differently. On MKRWAN1300 boards this pin was connected to the 3.3V rail. On MKRWAN13100 boards it seems to be connected to PB6/LPTIM1_ETR, presumably to allow the firmware enable or disable the internal TXCO IC. I am not sure what reason for this change is, perhaps power optimization?

Since I am not touching the configuration of the PB6/LPTIM1_ETR in the firmware, I wonder if the modem is not transmitting because the TXCO IC is powered off. All my modems (which are not MKRWAN based) have VDD_TCXO hardwired to 3.3V as well.

disk91, 2022-10-05:
Using PA12 is standard even if it's a problem when you want to use the USB port. 
I confirm your fix is resolving the transmission issue : radio signal seen on Spectrum analyzer and response from the AT+JOIN
```
23:03:04.407 -> +OK
23:03:04.407 -> 
23:03:12.165 -> +EVENT=1,1
```

disk91, 2022-10-05:
Thank you so much for the help.
I'll package a FW upgrade sketch for Arduino MKR1310 (your firmware is the way to select you own devEUI what Arduino do not support). Let me know if you plan a new release with this patch, so I'll update with it.
Paul

janakj, 2022-10-05:
Thanks for confirming. Yes, I will include this patch in the next release. But before I can do that, I will have to update my release scripts to generate two binary firmware variants. One for MKRWAN1310 with TCXO control on PB6 and debugging on USART2, and one for everything else.

disk91, 2022-10-06:
If anyone needs to deploy this FW on MKR1310, here is a fw loader project including the version above
https://github.com/disk91/mkr1310_openLoRaModem_fw_update

commit 568018e0 (2022-10-13): Add release variants for Arduion MKRWAN boards

Arduino MKRWAN boards only have USART2 connected, so the debugging
logger needs to be redirected there. MKRWAN1310 boards use PB6 to
control TCXO IC power instead of the default PA12.

Closes #109
Closes #108
Closes #107
closed 2022-10-13 by janakj