hardwario/lora-modem #110: Wisol

tcpipchip, 2022-10-16:
I will test this week in our wisol LoraWAN modules😀

tcpipchip, 2022-10-16:
[image]

tcpipchip, 2022-10-16:
got this error!
On WSL2

janakj, 2022-10-16:
You need to initialize git submodules. Please follow the steps in the section Building of README.md.

tcpipchip, 2022-10-16:
good morning!
Yes, i had done that but
needs key access rights
[image]

janakj, 2022-10-16:
Well, you need to be able to use Github from the VM where you are building it. If that's not the case, download the [source tarball from the most recent release](https://github.com/hardwario/lora-modem-abz/releases/download/v1.2.5/lora-modem-abz-1.2.5.tar.gz) and try to build that. The source tarball does not require working git/Github.

tcpipchip, 2022-10-16:
Now compiles! thanks!
Now i will start to convert to WISOL Module hardware!
It´s uses the STM32L071+SX1276
[image]

tcpipchip, 2022-10-16:
some tip to choose the target
STM32L072 during compile ?

janakj, 2022-10-16:
No idea. I am not familiar with that module. Also, you will need find out how the SX1276 chip is mapped on the GPIOs on the host MCU. This firmware only supports the mapping used by Murata Type ABZ. If your module uses a different GPIO mapping, you may need to modify the source code.

tcpipchip, 2022-10-16:
No problem! I will do! I was using the sdk at commands of stm on wisol. But looks that your is very better documented

tcpipchip, 2022-10-16:
> This firmware only supports the mapping used by Murata Type ABZ.

Only tell me where is the file mapping…

janakj, 2022-10-16:
src/main.c and src/sx1276-board.c.

I am going to close this issue since this is not a problem with the firmware. You are welcome to start a new discussion in the Discussions tab instead.

closed 2022-10-16 by tcpipchip

tcpipchip, 2022-10-17:
one suggestion is to make clear where is the NSS of SX1276 in the list files