hardwario/lora-modem #111: lom204

tcpipchip, 2022-10-17:
I got this message in wisol module
[image]
is it a progress ?

janakj, 2022-10-17:
Yes, that's progress. This event indicates that the module booted
correctly. You should be able to issue AT commands to it now.

tcpipchip, 2022-10-17:
yeah, i hope soon test  it if all going well we can include lom204 in the board list, right ? 
and i intend a blog :)

janakj, 2022-10-17:
Yes, we can include it in the list of supported boards if it works. Do you
have the datasheet or schematic for that board?

janakj, 2022-10-17:
Please use the Discussions tab for general questions like this. I would like to use Issues only for reporting bugs and problems with the software itself.

closed 2022-10-17 by janakj