hardwario/lora-modem #114: How to use Class C

disk91, 2022-10-19:
Hello, I'm trying to use the Class C mode
Is there something special to do other than calling the AT+CLASS=2 command ?

I don't see anything on the serial lien sending data from the network to the device ...
Do you have any sample code ?

Paul

disk91, 2022-10-19:
hum ... apparently the problem is upstream ... no radio emission from the gateway and it respond with an error message to the chirpstack submission ... so i'll back later with my question

janakj, 2022-10-19:
The network server needs to know that it can enqueue the downlink to the device immediately and also which gateway can reach the device. Make sure the network server knows the device is class C capable. Also, the device should have recently sent a Join, a LinkCheck, or an uplink message so that the network server can  learn which gateway can send downlinks to the device.

If you switch the device to class C without any prior uplink communication, the network server may not know where to enqueue downlinks to the device.