hardwario/lora-modem #115: Link check requests are being retransmitted

jefrin-altrosyn, 2022-10-26:
I am working on handling the behaviour of the abz module in the mkrwan 1310 board for cases where the device is not properly connected to the network due to issues like gateway down, out of range etc.

I made the device join the lora network and turned off the gateway after it sent few uplinks. When I send an link check command it fails but it is retransmitting the link check requests about 8 times which is my current RTYNUM count. In the wiki [here](https://github.com/hardwario/lora-modem-abz/wiki/AT-Command-Interface#atlncheck) it is mentioned that the link check requests are not retransmitted and will be sent only once per request. 

[image]

[image]

1. Is this link check behaviour normal or did I misunderstood something?
  
2. This is not related to the above issue but do you have any idea on how often should (or when) the modem be made to re join the network?

janakj, 2022-10-26:
Thank you for the report! Link check requests transmitted without application payload should not be retransmitted indeed. From your logs it seems that I made a mistake here. The modem should ignore the AT+REP value in this case which does not appear to be the case. I will check this.

It's hard to say how often the modem should rejoin in general. That depends on many factors, e.g., physical parameters of your network, the number of devices, intended application, security requirements, etc. The main benefit of rejoining periodically is that your session keys get reset. That limits the total number of bytes encrypted with the same session key which is good from a security perspective.

On the other hand, it may be difficult to execute a Join automatically and reliably by devices deployed in the field. An OTAA Join is a fairly disruptive operation from the device perspective. It resets many MAC parameters, including the parameters negotiated with ADR. In addition, this implementation transmits OTAA Joins with DR0 by default (for reliability), which can drain the battery if performed too often.

janakj, 2022-11-12:
Hi @jefrin-altrosyn, I am not able to reproduce the link check retransmissions. Are you still able to reproduce the behavior? Also, what values do you have in `AT+REP` and `AT+RTYNUM`?

commit 45820478 (2022-11-12): Disable link check retransmissions

Link check requests issued via AT+LNCHECK and sent in standalone empty
uplinks should not be retransmitted, regardless of the AT+REP value.

Since the originaly implementation did not reconfigure the
ChannelsNbTrans MIB value, a retransmitted uplink sent prior to the link
check uplink would trigger link check retransmission. This patch fixes
the problem by always resetting the retransmission counter before
sending a link check request.

Closes #115
closed 2022-11-12 by janakj