hardwario/lora-modem #116: Feature request - add put/get command to store data in NVM

disk91, 2022-10-26:
Could you add AT+NVM=a,0xVV and AT+NVM?a where a is an address (0 to 16 or 32) and VV is a 8b hex value. 

This command will allow to store and read data in the murata NVM space. One of the use is to store a modem setting state and some extra information. 

It is usefull when the nvm capabilities of a board is limited (like mkr1310 - even if it is supposed to have an onboarded spi flash)

janakj, 2022-10-26:
I would be happy to consider a pull request if you want to submit one. The command should use the `AT$` prefix (`AT+` is reserved for Murata-compatible commands).

disk91, 2022-10-26:
Unfortunately have so many projects pending :(
I will see if I found some time for doing it

janakj, 2022-10-27:
Please feel free to start a new discussion in the Discussions tab if you start working on it and need advice.

disk91, 2022-10-27:
Having some train today and setup the environment, will see if I can quickly made an addition for this

disk91, 2022-10-27:
I'm close to a working version.
the current limit is the EEPROM available size, I can have a maximum of 4 byte of user data.
The code is only using 3KB on the 6KB available and I assume most of the partitions are over-estimated.
do you think @janakj we can reduce some of the partitions to extend the User NVM space ?

disk91, 2022-10-27:
merge request sent

janakj, 2022-10-28:
> I'm close to a working version. the current limit is the EEPROM available size, I can have a maximum of 4 byte of user data. The code is only using 3KB on the 6KB available and I assume most of the [...]

Existing partitions are left slightly larger than necessary to accommodate for future additions. Also, the size of some of LoRaMac-node data structures stored in NVM changes depending on what features and regions are enabled at compile time.  I sized the existing partitions to almost entirely fill one EEPROM bank (half total EEPROM size). But we can definitely carve out more space for the user data partition. There should be plenty of space.

I will see what can be resized when I get around to reviewing your pull request.

commit ee946a58 (2022-11-13): Add AT$NVM command to store user data in NVM

The command AT$NVM makes a portion of the NVM (EEPROM) in the Type ABZ
module available to the application. The memory is available in the form
of 64 single-byte registers. The application can read and write the
values in each register individually.

Use the syntax AT$NVM 0,123 to store value 123 in NVM register 0. Use
the syntax AT$NVM 0 to read the value previously stored in NVM register
0: +OK=123.

Each NVM register can store a single byte (value 0 to 255). The values
are encoded in decimal in the ATCI. Currently, the firmware exports 64
NVM re [... cut]
closed 2022-11-13 by janakj