hardwario/lora-modem #130: Default delay values

ctorney, 2023-04-01:
Hi,
I have a question about the RX delay and join delay values. 

In the documentation it says:
**The times are encoded into a comma-delimited value as follows: <RX1>,<RX2>,<JoinAccept RX1>,<JoinAccept RX2>.** 
and
**The default value upon factory reset is 5000,6000,1000,2000.**

However in this issue https://github.com/hardwario/lora-modem/issues/24 it seems that the default value should be 1000,2000,5000,6000

janakj, 2023-04-02:
Hi @ctorney,

Thanks for pointing this out! The factory default value "5000,6000,1000,2000" is correct. We try to remain compatible with the original Murata Modem firmware and this is the default value that firmware uses in version 1.1.06:
```
AT+VER?
+OK=1.1.06,Aug 24 2020 16:11:57

AT+DELAY?
+OK=5000,6000,1000,2000
```
The default delay values in #24 are wrong.

-Jan

ctorney, 2023-04-02:
Hi @janakj 
Maybe I'm misunderstanding but it seems from the code here
https://github.com/hardwario/lora-modem/blob/698b503b7e486aa2e922d06a36d790f72682fade/src/cmd.c#L822
that the delay values in https://github.com/hardwario/lora-modem/issues/24 are correct but the documentation should be list the values in the order:
\<JoinAccept RX1\>,\<JoinAccept RX2\>,\<RX1\>,\<RX2\>

janakj, 2023-04-02:
You are right, this is odd. I have reopened the ticket and will investigate this coming week. Thanks for bringing this up!

-Jan

janakj, 2023-05-20:
Hi @ctorney,

Thank you so much for reporting this! You were right; the documentation was incorrect. The delay values are encoded in the following order: JoinAccept window 1, JoinAccept window 2, Receive window 1, Receive window 2. I have updated the [wiki page](https://github.com/hardwario/lora-modem/wiki/AT-Command-Interface#atdelay). I also cross-checked this with the original TypeABZ firmware.

-Jan

closed 2023-05-20 by janakj