hardwario/lora-modem #131: Modem firmware update by bcf command

zokl, 2023-04-27:
Hello  @janakj, 

based on the tutorial at https://github.com/hardwario/lora-modem/wiki/LoRa-Module-Firmware-Update I made an FW update with an unmodified version of 'bcf' app. I don't need your modified version. I need only the right wire connection directly to the module (service pins). I think that for tutorials will be better to use the commonly used APP which is available in the release trunk.

janakj, 2023-04-27:
Did you wire everything up as shown in the diagrams, i.e., BOOT and RESET to DTR and RST, or did you use another approach to control BOOT and RESET?

zokl, 2023-04-27:
> Did you wire everything up as shown in the diagrams, i.e., BOOT and RESET to DTR and RST, or did you use another approach to control BOOT and RESET?

I wire it in the same way that you draw but instead of P2 and P3 I use pins on service connector.

janakj, 2023-04-28:
So you have it connected differently then, not according to the wiki (the wiki tutorial connects RESET to RST). I haven't tried this, but maybe this simply works by chance with an unmodified bcf version. The bcf tool implements a boot and reset sequence for the Core Module which has these signals wired differently.

I'll swapping RTS and DTR on my Lora modules to see if it works. If it does, that's probably a better way to implement this since then the PR for -m LoRa would no longer be needed.

-Jan

janakj, 2023-05-06:
You were right! With swapped RTS/DTR signals, one can indeed use an unmodified bcf version. That's great because I can drop the PR and the whole process is going be simpler. Thanks for the suggestion! I have updated the wiki.

-Jan

closed 2023-05-06 by janakj